The-EG/OctoPrint-UBLMeshEdit #1: [FR] Save mesh to file

lightmaster, 2021-02-24:
Would it be possible to save and download the mesh data to a file? I like to copy/paste the data into a text file just to be safe, and it would be much easier if its automated with this plugin. Could also implement restoring a saved mesh from a file too.

The-EG, 2021-02-24:
Yes, that is already on the 'todo' list. I think I'd prefer a simple CSV format, but I'm open to input.

lightmaster, 2021-02-24:
Oh yea, a txt file would not be ideal for a script/plugin to read and write from, just meant that's how I'm doing it right now, just copy/pasting the topology map.

Why not use a JSON file? More rigid structure to it, with proper labels its also fairly human readable too, and most methods of reading and writing a JSON file are more efficient than a CSV file.

The-EG, 2021-02-24:
> Why not use a JSON file? More rigid structure to it, with proper labels its also fairly human readable too, and most methods of reading and writing a JSON file are more efficient than a CSV file.

Good point! I'll keep that in mind.

The-EG, 2021-02-25:
I decided to go a bit of a different route: I have added an 'Export GCode' button. This will download a GCode script that will restore the current mesh when run. To restore the mesh, just load it like a normal file and 'print' it in OctoPrint. 

I went this route because restoring the mesh is essentially just running these same commands, and this ends up being simpler (I don't have to parse the file and build the commands, etc).

The current caveat of 'only works with valid meshes' applies to this as well. I'll have to revisit when/if I handle invalid points better in the future.

If you'd like to test it out before I make a RC/release, you can install the branch directly using this url: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/feature/export_gcode.zip

You can then always go back to the release version by reinstalling using: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/main.zip

lightmaster, 2021-02-25:
> The current caveat of 'only works with valid meshes' applies to this as well.

So the mesh to be exported must have all spots with valid offsets? Even with the circular bed of a Delta, your mesh should be filled with all valid offsets since those outside the bed are used to extrapolate the offsets around the edge of your bed.

The-EG, 2021-02-25:
> So the mesh to be exported must have all spots with valid offsets? Even with the circular bed of a Delta, your mesh should be filled with all valid offsets since those outside the bed are used to extrapolate the offsets around the edge of your bed.

Yes. In general, I haven't added checks for invalid values, so the plugin trying to operate on them would be 'undefined behavior.' In this case it would probably fail to set that point when running the gcode and the print would then stop (not trying the rest).

I need to understand the process a bit better before I try to approach that though...it's really sounding like an invalid mesh shouldn't be something that would be saved anyway, so it could be as simple as not allowing the export if the mesh isn't valid, etc.

Also, if you happened to try the branch before or while I was fixing the error, it's now fixed (the exported mesh was mirrored vertically).

lightmaster, 2021-02-25:
Haven't had a chance to test it yet.

As far as exporting invalid data (which shouldn't really happen anyways, but you know Murphy's Law...), I assume you're using a loop to extract data, reformat to `M421` command, rinse and repeat. How about adding a simple `if data is invalid, then skip the rest of this iteration`. Then the exported GCode won't contain any invalid data and should restore without issue.

Good data:
```
G29 T1

Bed Topography Report for CSV:

0.073	0.000	0.000	0.025	0.115	0.170	0.070	0.034	0.034	-0.100
0.060	0.030	0.015	0.035	0.016	0.066	0.040	0.035	-0.100	-0.100
0.047	0.055	0.023	0.037	0.014	0.040	0.005	-0.004	-0.095	-0.100
0.260	0.115	0.069	0.037	0.021	0.023	-0.001	-0.011	-0.090	-0.215
0.300	0.179	0.087	0.048	0.018	0.007	-0.010	-0.034	-0.097	-0.220
0.365	0.225	0.125	0.055	0.027	0.011	0.009	-0.031	-0.084	-0.225
0.390	0.225	0.144	0.052	0.039	0.009	-0.001	-0.038	-0.075	-0.225
0.236	0.195	0.104	0.028	0.026	-0.001	0.008	-0.025	-0.035	-0.035
0.330	0.170	0.020	-0.025	-0.013	-0.023	0.010	0.025	0.005	0.005
0.424	0.045	0.010	-0.025	-0.090	-0.050	0.012	0.070	0.129	0.045
ok
```

Invalid data:
```
G29 T1

Bed Topography Report for CSV:

NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
ok
```

So something as simple as `if I0 J0 = "NAN", then skip`. I don't know Javascript, so not sure how to code that correct in JS, but you get the point.

Side note, how are you extracting the data from the firmware without making it beep a couple time in acknowledgement? Every time I send `G29 T1`, I get a couple beeps back from the printer, but with your plugin its silent.

lightmaster, 2021-02-25:
Exported mesh looks good to me. About the filename, can we append a date/time stamp to it? `Restore Mesh 2021-02-25 17-30.gcode`? Specifically that format since it will automatically sort itself by year, month, day.

lightmaster, 2021-02-25:
I invalidated my entire mesh, "printed" the `Restore Mesh.gcode` file, then compared the resulting mesh and my original mesh. Both the restored mesh and the original mesh were 100% identical.

The-EG, 2021-02-25:
> How about adding a simple if data is invalid, then skip the rest of this iteration

Yep, you have it right and that's the idea. But, there are multiple points it can cause issues at, on both the Python side and Javascript side, so I really need to dig into it, and create some good test cases (output for the Virtual Printer has to be hand crafted for things like this). Then I can address it properly; I suspect the mesh buttons/save point will need some similar treatment too.

> Side note, how are you extracting the data from the firmware without making it beep a couple time in acknowledgement? Every time I send G29 T1, I get a couple beeps back from the printer, but with your plugin its silent.

I use `M420 V1 T1` to get the current mesh and `M421` to set the point since they don't trigger the buzzer. I guess some mike like the feedback, but I found it to be a bit annoying especially considering how loud the buzzers are on my Ender 3's. They make a quick 'chirp' but it's still audible half way across the house...

Unfortunately, the clear mesh, save and load to slot buttons still use `G29`, so they will cause a beep/chirp but at least those aren't generally something that you'll be pressing multiple times in succession.

The-EG, 2021-02-25:
> About the filename, can we append a date/time stamp to it? `Restore Mesh 2021-02-25 17-30.gcode`? Specifically that format since it will automatically sort itself by year, month, day.

Now that you say that, I may as well make it configurable:
[image]

The date format is hard coded, but I think it's sensible enough.

Thanks for testing and providing input! I'll get this pushed out to an RC soon.

The-EG, 2021-02-25:
This is now available in v0.2.0-rc1.

Anyone who would like to try this out before official release may do so by selecting the 'Release Candidate' channel for this plugin:
[image]

closed 2021-02-25 by The-EG

lightmaster, 2021-02-28:
Thanks for this. Gonna be using this in the next few days to move my mesh to the SKR 1.4 Turbo board whenever it arrives. Saves a lot of trouble having to copy/paste and then reformat to import the mesh.