The-EG/OctoPrint-UBLMeshEdit #2: [FR] Send GCode commands to go to mesh spot, automating fine tuning a mesh.

lightmaster, 2021-02-24:
When using UBL without a LCD with an encoder wheel, you must manually enter the commands to go to each spot to fine tune the probing. I propose a toggle switch that enables and disables automatically going to the mesh spots when you click on the boxes to edit them. Due to the Z-Height speed difference between Deltas and Cartesian style printers, a Z Feedrate selector would be nice too.

With the proposed step \#4 under `Once User clicks on a square`, this Feature Request would also allow this plugin to be used to create a manual mesh from scratch as well as fine tuning an existing mesh.

### User clicks on button to enable this new feature:
1. Check that bed and nozzle temp are within a reasonable temp range. If not instruct user to heat them up first. Octoprint Printer Profile has a checkbox for whether or not a heated bed is present, so I assume that this can be checked so we don't error out for users without a heated bed.
2. Prompt user for confirmation before continuing (wanna make absolutely sure they are ready before any motion happens)
3. Send GCode `M18 S3000` to keep stepper motors from disabling while probing.
4. Send GCode `M28` to home
5. Send GCode `G1 Z2 Fxxx` to get close to bed plate (vital for Deltas since they can not move horizontally until they have descended at least some, might as well get close to bed plate at this point too)
6. Send GCode `M211 S0` to disable software endstops so user can go Negative if needed.
7. Wait until user picks a spot to go to.

### Once User clicks on a square:
1. Send GCode `G42 Ix Jy` to move to the chosen spot
2. Send GCode `G1 Z0.4 Fxxx` to get as close to bed plate as is safe.
3. Wait for User to manually probe bed.
4. Send GCode `M421 Ix Jy Q/Zx.xxx` to set value of the spot. If the spot currently has no value, send `Zx.xxx` to set the absolute value, else if the spot had an existing value, send `Qx.xxx` to add or subtract from the existing value.
4.1. If user entered a value into the box, use the value they entered. If they left box blank send and check `M114` to get current position and use that value automatically.
5. Send GCode `G1 Z2 Fxxx` to get back off the bed. This will also automatically take the new adjustment into effect so User can repeat this process to verify their adjustment.

### If User clicks on a 2nd square without saving an adjustment:
1. Send GCode `G1 Z2 Fxxx` to get back off the bed.
2. Continue with the steps under `Once User clicks on a square`

### Once User clicks the button to disable this new feature:
1. Send GCode `G29 Sx` to save mesh.
2. Send GCode `G28` to get back home.
3. Send GCode `M18 S300` to set stepper motor timeout to 5m. Not sure of an easy way to get this value to be able to revert it, so I feel that a 5m timeout without any motion is reasonable. Obviously adjust this value if you disagree.
4. Send GCode `M18` to disable stepper motors now.

The-EG, 2021-02-24:
This one is also already on the planned to-do list and something I definitely plan to implement at some level. 

To be honest, I *just* started using UBL about 30 minutes before I started writing this plugin, so I'm still working out the process myself and what you've laid out is helpful.

I do envision 2 'modes' for this, at least. One, like you've described, would be for building a mesh, or recalibrating multiple points by manually probing, similar to the LCD menu, but through the plugin. The other, would be more of a 'show me where on the bed this point is,' essentially just `G42` (another user requested this on Discord earlier).

In general, instead of having hard coded scripts for the start and end ('user clicks the button to enable', etc) I'd opt to make that configurable with some sensible defaults.

On a different note: are you using this on a delta printer? I only started with square meshes in mind, so I have no idea how this will function on anything else. I'd be interested in how it's working, and outputs of `M420 V T1` if it's not working properly.

lightmaster, 2021-02-24:
In the middle of switching nozzles right now, so haven't actually used the plugin besides to view the current mesh, which worked good.

As far as command differences, what I described above is a generalized version of what I do to start a new mesh or fine tune one, and should be generalized to work with any style printer.

The only real difference between Delta and Cartesian as far as UBL is concerned is the fact that Deltas can't physically probe all the points that UBL uses. For instance, with my 100 point setup, in column `I0`, the points that are physically over my bed are `J4 J5`, however it still uses `J0 J1 J2 J3 J6 J7 J8 J9` to extrapolate that value that would be in between points along the edge of the bed. These points that are not located on the bed are set after the bed's points are set, using the command `G29 P3` to automatically calculate and populate the invalid points that are immediately adjacent to the most outwardly valid points. This means that multiple uses of `G29 P3` are usually needed to fill in the full square bed for the mesh.

Hope that makes some sense, if not I'll try to explain it better with pictures, lol.

As far as how this plugin will work with Deltas, I'd suggest showing the 0-indexed column and row headers, as well as the corresponding X,Y coordinates for those headers. This way Delta users can see and know which points have a physical location on their beds and which ones lay outside their beds. Delta users could also just click a square and see where the nozzle moves to, then click the next outwardly one and see if it moves further outward, or if it remains still. Perhaps a way to manually desaturate the squares that lay outside the bed so a Delta user could visually see which ones physically exist? Or maybe be able to outline the outside-the-bed squares in red?

The-EG, 2021-03-03:
> When using UBL without a LCD with an encoder wheel, you must manually enter the commands to go to each spot to fine tune the probing.

> 3. Wait for User to manually probe bed.

I'm curious, without an encoder do you still manually send commands to make the z adjustments or do you have some other physical buttons on the printer or something?

lightmaster, 2021-03-03:
I use the TFT24 touch screen to adjust it up and down by 0.025mm per click, then make a `M421 Q` offset, go back up to 0.2mm and verify the offset. Repeat as needed then progress to the next spot.

The annoyance is without the encoder wheel, I must do all the commands manually. I can initiate it to semi automate the fine tuning, however there's no way to move the nozzle up and down during that without the encoder wheel. TBH, no idea why they don't just add a second set of buttons to the screen to be able to move it and then set the offset

The-EG, 2021-03-03:
My WIP for this is now in the feature/interactive_probing branch.

In general, my philosophy is that the user bears some responsibility for knowing their printer and configuration. To that end, I think having things configurable with reasonable defaults is the best way to provide functionality to most users that 'just works' while still supporting use cases that may be less common.

So this feature, which I'm calling 'Interactive Probing', will have multiple configuration options. So far:
 - Pre GCode: gcode that is run when the mode is enabled
 - Post GCode: gcode that is run when the mode is disabled
 - Feed rates (z & x/y in mm/s)
 - 'Z Hop': what height should the hotend be raised to for each move?
 - 'At Point Z': how low should the hotend be lowered to once the move is complete?

For example, rather than trying to figure out if the printer has a heated bed (easy enough), and possibly showing a warning if it's not heated up, which would also mean that I'd have to have a configuration value for that minimum temp and an option to not show the warning: a user that wants to heat the bed prior to using Interactive Probing can add the appropriate command(s) to the Pre GCode setting.

Providing the movement is fairly straight forward (click for video):
[image]

I'll have to give some more thought to how I want to go about handling editing the point. I'm also not sure that I like how I have the pre and post gcode running.