The-EG/OctoPrint-UBLMeshEdit #3: [FR] support of Automatic Bed Leveling (ABL)

KlausMu, 2021-02-25:
Would it be possible to also support ABL and not only UBL? As far as I understand it should only cause minor changes in the code. 

### EEPROM slots
In ABL there is no support for several EEPOM slots. You could simply replace this with M500/M501.

### phases
Phases are not supported in ABL. You are using G29 P0, which could be replaced with
- M421 Ix Jy Z0.0
- M420 S0

The-EG, 2021-02-25:
Viewing and editing the mesh should work for ABL, in theory. 

I worded the disclaimer pretty strongly because I do not use ABL, so I couldn't test it directly. The mesh is retrieved with `M420 V1 T1` and each point is set with `M421 Ix Jy Zz` as you mentioned. 

Can you please run `M420 V1 T1` and attach the output?

As long as ABL supports the `T1` parameter (the Marlin docs don't say which modes supports which `T`) and it's not vastly different from UBL output, I believe I can detect the difference and show a 'Save/Load EEPROM' buttons instead of from slots. The plugin will still be called 'UBL,' though!

If that is the case, it's a fairly minor change that I'd be happy to make.

KlausMu, 2021-02-25:
Thanks for your fast reply. Here is the output.

Send: M420 V1 T1
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -1.565 -1.592 -1.590 -1.567 -1.587
Recv:  1 -1.667 -1.682 -1.617 -1.562 -1.410
Recv:  2 -1.770 -1.742 -1.647 -1.550 -1.440
Recv:  3 -1.860 -1.790 -1.680 -1.552 -1.402
Recv:  4 -1.927 -1.817 -1.695 -1.530 -1.352
Recv: 
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok

The-EG, 2021-02-27:
This is now in the `feature/abl_compat` branch. If you'd like to try it you can reinstall (no uninstall needed) the plugin using this URL: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/feature/abl_compat.zip

You can return to the stable version by reinstalling from the main URL: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/main.zip

A few assumptions are made, namely that the J values are swapped from UBL (I0, J0 = lower left corner of the bed for UBL, upper left for ABL). 

A prominent warning will be displayed if a non-ubl mesh is detected. I'll add a setting to hide this before release.

KlausMu, 2021-02-27:
I did a test and it worked very fine. Thanks a lot!

closed 2021-02-27 by KlausMu

The-EG, 2021-02-27:
> I did a test and it worked very fine. Thanks a lot!

Thank you. I'll get this into a release candidate soon, and then a stable release sometime after that.