The-EG/OctoPrint-UBLMeshEdit #4: cant see and edit the mesh

simsemas, 2021-02-25:
the Program does not load the existing mesh properly to edit it. 
I Have a saved one and the tool sends the proper commands (g29 T and g29 L1) but it does not display any mesh in the tool. 
In terminal i get the mesh displayed and it is valid.

The-EG, 2021-02-25:
What version of Marlin are you running, and why style of printer is this (cartesian/delta)?

Please attach the output of `M420 T1 V1`. 

> I Have a saved one and the tool sends the proper commands (g29 T and g29 L1)

This plugin doesn't currenty use `G29 T`, so if you are seeing that command it's from something else (it uses `M420 V1 T1` instead).

simsemas, 2021-02-25:
Hi! Thx for your reply. 

Its Marlin 2.x. dont know which one exactly on a cartesian printer 

M420 T1 V1 give the following results. 

Send: M420 T1 V1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: 0.307	0.185	0.198	0.440	0.351	0.403
Recv: 0.059	0.052	0.107	0.285	0.274	0.340
Recv: 0.250	0.180	0.122	0.431	0.349	0.311
Recv: 0.293	0.148	0.204	0.468	0.491	0.504
Recv: 0.385	0.236	0.231	0.496	0.393	0.487
Recv: 0.422	0.308	0.319	0.547	0.483	0.520
Recv: Mesh is valid
Recv: Storage slot: 1
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 10.00
Recv: ok P15 B15Send: M420 T1 V1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: 0.307	0.185	0.198	0.440	0.351	0.403
Recv: 0.059	0.052	0.107	0.285	0.274	0.340
Recv: 0.250	0.180	0.122	0.431	0.349	0.311
Recv: 0.293	0.148	0.204	0.468	0.491	0.504
Recv: 0.385	0.236	0.231	0.496	0.393	0.487
Recv: 0.422	0.308	0.319	0.547	0.483	0.520
Recv: Mesh is valid
Recv: Storage slot: 1
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 10.00
Recv: ok P15 B15

The-EG, 2021-02-25:
Ok, I think I see the issue with this one. Should be an easy fix.

The-EG, 2021-02-25:
If you're willing to help test this before I push it to a release, you can install using this URL: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/fixes.zip

You can go back to the release afterwards by reintalling using the main URL: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/main.zip

simsemas, 2021-02-25:
Hi i installed the test version. 
Still the same problem...

Send: M420 T1 V1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: 0.307	0.185	0.198	0.440	0.351	0.403
Recv: 0.059	0.052	0.107	0.285	0.274	0.340
Recv: 0.250	0.180	0.122	0.431	0.349	0.311
Recv: 0.293	0.148	0.204	0.468	0.491	0.504
Recv: 0.385	0.236	0.231	0.496	0.393	0.487
Recv: 0.422	0.308	0.319	0.547	0.483	0.520
Recv: Mesh is valid
Recv: Storage slot: 1
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 10.00
Recv: ok P15 B15
[...]

andreibosco, 2021-02-25:
I'm also having the same issue. I'm running Marlin 2.0.7.2 and here's the output of `M420 T1 V1`:
```
Send: M420 V1 T1
Recv: Mesh Bed Level data:
Recv: 5x5 mesh. Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2        3        4
Recv:  0 +0.10000 -0.00000 -0.07500 -0.05000 +0.02500
Recv:  1 +0.02500 -0.02500 -0.05000 -0.02500 +0.05000
Recv:  2 -0.00000 -0.00000 -0.05000 -0.00000 +0.12500
Recv:  3 +0.05000 -0.00000 -0.02500 +0.00000 +0.05000
Recv:  4 +0.05000 +0.00000 -0.05000 +0.02500 +0.10000
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 2.00
Recv: ok
```

The-EG, 2021-02-25:
> Hi i installed the test version.
> Still the same problem...

@simsemas 
Argh, ok, I missed the additional output after `ok`.  Sorry about that. If you installing the fixes URL again it should work.
Out of curiosity, is this a firmware you build or is it precompiled from somewhere? This is a bit different than what I've seen for vanilla Marlin.

> I'm also having the same issue.

Hi @andreibosco , unfortunately this plugin doesn't support Mesh Bed Leveling (MBL), which is what you have enabled (I can tell by the line that says `Mesh Bed Level data:`). If you are building firmware yourself, I'd recommend giving UBL a try. Since you are using MBL already, you'd want to switch to UBL and use `PROBE_MANUALLY`. The commands are a bit different, but the concept is the same, and UBL has quite a bit more functionality.

simsemas, 2021-02-25:
the firmware i use is this one https://github.com/Vertabreak/Vert-Marlin. its a great pice of art and especially made for geeetech printers. 

i installed the fixes URL again and now it seems to work. 

I get the displayed results and can change them. the changes are also saved. 

So everything works as intended. 
Thank you very much for the tool and the support! 

Pls close

closed 2021-02-25 by simsemas

The-EG, 2021-02-25:
This fix has been released to v0.1.1

andreibosco, 2021-02-25:
> Hi @andreibosco , unfortunately this plugin doesn't support Mesh Bed Leveling (MBL), which is what you have enabled (I can tell by the line that says `Mesh Bed Level data:`). If you are building [...]

Thanks a lot for the tip, I will compile a new build of marlin and try it out.

The-EG, 2021-02-25:
> Thanks a lot for the tip, I will compile a new build of marlin and try it out.

If you want more info, the marlin doc on it goes into quite a bit of depth: https://marlinfw.org/docs/features/unified_bed_leveling.html

Since you have a mesh already, rather than starting from scratch and manually probing each point again, you'd probably want to just enable it, clear the mesh (start with all 0s) and then just manually enter your old MBL mesh z values back in (that's more or less what this plugin is for!)