The-EG/OctoPrint-UBLMeshEdit #5: It is possible to include Bilinear Levelling? Looks much the same.

tigercjn, 2021-03-01:
[image]

It is for the Snapmaker 2.0 series printer.

Send: M420 V1 T1
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +6.846 +6.733 +6.764 +6.691 +6.696
Recv:  1 +7.058 +7.153 +7.148 +7.083 +6.885
Recv:  2 +7.165 +7.229 +7.153 +7.118 +6.914
Recv:  3 +7.070 +7.160 +7.125 +7.041 +6.890
Recv:  4 +6.843 +6.955 +6.926 +6.880 +6.808
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +6.84625 +6.80297 +6.75968 +6.73250 +6.73602 +6.75565 +6.76375 +6.74440 +6.71352 +6.69125 +6.68718 +6.69171 +6.69625

The-EG, 2021-03-01:
It might be possible; I'm not very familiar with ABL but this looks like it's not quite the normal Marlin ABL report either.

Please include the full output of `M420 V1 T1` all the way until 'ok' is received.

Also, is this a custom/vendor version of Marlin or something you built? From just the snippet you included, it looks like it's doing some sort of subdivision on the mesh. I don't see that documented in Marlin, so I can't say what will happen if you edit the mesh with this plugin (does the subdivision automatically update? does some other command need to be run?).

tigercjn, 2021-03-02:
The simple change so it shows the mesh and edit a point would be great.
Here is a full printout.

Send: M420 V1 T1
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +6.846 +6.733 +6.764 +6.691 +6.696
Recv:  1 +7.058 +7.153 +7.148 +7.083 +6.885
Recv:  2 +7.165 +7.229 +7.153 +7.118 +6.914
Recv:  3 +7.070 +7.160 +7.125 +7.041 +6.890
Recv:  4 +6.843 +6.955 +6.926 +6.880 +6.808
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +6.84625 +6.80297 +6.75968 +6.73250 +6.73602 +6.75565 +6.76375 +6.74440 +6.71352 +6.69125 +6.68718 +6.69171 +6.69625
Recv:  1 +6.92051 +6.90669 +6.89286 +6.88523 +6.89131 +6.90357 +6.90570 +6.88881 +6.86178 +6.83486 +6.81153 +6.78831 +6.76509
Recv:  2 +6.99477 +7.01041 +7.02605 +7.03796 +7.04659 +7.05150 +7.04764 +7.03322 +7.01003 +6.97847 +6.93588 +6.88491 +6.83394
Recv:  3 +7.05750 +7.09287 +7.12824 +7.15250 +7.16046 +7.15732 +7.14750 +7.13519 +7.11621 +7.08250 +7.02648 +6.95574 +6.88500
Recv:  4 +7.10852 +7.14677 +7.18501 +7.20875 +7.20918 +7.19511 +7.17843 +7.16827 +7.15550 +7.12468 +7.06462 +6.98650 +6.90838
Recv:  5 +7.14801 +7.17940 +7.21079 +7.22681 +7.21648 +7.19079 +7.16727 +7.15891 +7.15272 +7.12727 +7.06901 +6.99150 +6.91398
Recv:  6 +7.16500 +7.19144 +7.21787 +7.22875 +7.21218 +7.18005 +7.15250 +7.14403 +7.14014 +7.11750 +7.06209 +6.98792 +6.91375
Recv:  7 +7.15324 +7.18114 +7.20905 +7.22162 +7.20789 +7.17882 +7.15209 +7.13963 +7.12949 +7.10347 +7.05021 +6.98105 +6.91190
Recv:  8 +7.11898 +7.15025 +7.18152 +7.19838 +7.19198 +7.17117 +7.14806 +7.12970 +7.10906 +7.07709 +7.02701 +6.96561 +6.90421
Recv:  9 +7.07000 +7.10463 +7.13926 +7.16000 +7.15940 +7.14491 +7.12500 +7.10320 +7.07597 +7.04125 +6.99584 +6.94292 +6.89000
Recv: 10 +7.00398 +7.04132 +7.07866 +7.10176 +7.10290 +7.08979 +7.07144 +7.05089 +7.02508 +6.99380 +6.95514 +6.91100 +6.86685
Recv: 11 +6.92324 +6.96328 +7.00331 +7.02838 +7.02971 +7.01608 +6.99884 +6.98200 +6.96155 +6.93690 +6.90646 +6.87182 +6.83718
Recv: 12 +6.84250 +6.88523 +6.92797 +6.95500 +6.95653 +6.94236 +6.92625 +6.91310 +6.89801 +6.88000 +6.85778 +6.83264 +6.80750
Recv: 
Recv: leveling ON
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok

The-EG, 2021-03-02:
It does appear to be a fairly simple fix. If you'd like to try it, install the plugin (no un-install necessary) from this URL: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/fixes.zip

You can return to the release version by installing from the main URL at any time: https://github.com/The-EG/OctoPrint-UBLMeshEdit/archive/main.zip

tigercjn, 2021-03-03:
Sorry, I missed your extra question, it is just the Marlin used in the Snapmaker series of printers, CATMULL ROM Leveling Grid has been around for a long time I would say as there are bug reports back as 2017 on the Marlin GitHub page.

tigercjn, 2021-03-05:
What can I say, thank you, it simply works now, a great plugin.
I changed a few points and they changed fine and are reflected in the terminal readout.
Well Done.
Cheers
Chris

The-EG, 2021-03-05:
Thanks! This is now in 0.2.2.

closed 2021-03-05 by The-EG