The-EG/OctoPrint-UBLMeshEdit #6: [BUG] The mesh doesn't display

Gaara14, 2021-03-01:
Hello, thanks for this plugin. I have installed UBL to use it.

I'm using Octoprint Version 1.5.3 on Raspberry Pi 3 B
Printer: Tevo Tornado, Marlin from Tevo repository (1.x)
Sensor: BLTouch

Bug: The "Get Mesh" button has no graphical effect, even if I can see the M420 V1 T1 result.

**M420 Output**
Send: M420 V1 T1
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,9)                                                                   (9,9)
Recv: (5,295)                                                                (315,295)
Recv:      .          .          .          .          .          .          .          .          .          .
Recv: 
Recv:   0.059   -0.003   -0.011   -0.051    0.007   -0.011    0.004   -0.036       .          .
Recv: 
Recv:   0.079    0.019   -0.013   -0.071   -0.051   -0.041   -0.033   -0.053       .          .
Recv: 
Recv:  -0.003   -0.043   -0.091   -0.101   -0.104  [-0.088]  -0.026   -0.061       .          .
Recv: 
Recv:   0.027   -0.033   -0.086   -0.028    0.007   -0.058   -0.063   -0.093       .          .
Recv: 
Recv:   0.034    0.012   -0.076   -0.088   -0.086   -0.068   -0.031   -0.011       .          .
Recv: 
Recv:   0.012   -0.021   -0.056   -0.106   -0.068   -0.053   -0.018    0.012       .          .
Recv: 
Recv:   0.079    0.039    0.044   -0.023    0.014   -0.043 T:18.39 /0.00 B:18.40 /0.00 @:0 B@:0
Recv:    -0.011   -0.036       .          .
Recv: 
Recv:   0.119    0.082    0.069    0.032   -0.028   -0.051   -0.023   -0.071       .          .
Recv: 
Recv:   0.159    0.112    0.087    0.052   -0.008   -0.013   -0.036   -0.078       .          .
Recv: (5,5)                                                                    (315,5)
Recv: (0,0)                                                                     (9,0)
Recv: UBL_MESH_VALID = 1
Recv: ubl.state.storage_slot = 1
Recv: echo:Bed Leveling Marche
Recv: echo:Fade Height 10.00
Recv: ok

**Firmware**
printer.firmware: Marlin TORNADO (Github)

[...]
#elif ENABLED(AUTO_BED_LEVELING_UBL)

  //===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  #define UBL_MESH_INSET 5          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Was 10 Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_PROBE_PT_1_X 39       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 180
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 30
  #define UBL_PROBE_PT_3_X 180
  #define UBL_PROBE_PT_3_Y 20

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION (MIN_PROBE_EDGE)
  #define RIGHT_PROBE_BED_POSITION (X_BED_SIZE + X_PROBE_OFFSET_FROM_EXTRUDER - MIN_PROBE_EDGE)
  #define FRONT_PROBE_BED_POSITION (MIN_PROBE_EDGE)
  #define BACK_PROBE_BED_POSITION (Y_BED_SIZE + Y_PROBE_OFFSET_FROM_EXTRUDER - MIN_PROBE_EDGE) 

  #define UBL_G26_MESH_VALIDATION   // Enable G26 mesh validation
  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle

#elif ENABLED(MESH_BED_LEVELING)
[...]

What I'm doing wrong ?
Thanks

The-EG, 2021-03-01:
Wow, that is quite different from the standard Marlin 2.0.x output. Sounds like this firmware is based on Marlin 1.x, not 2.x, so that could be where quite a bit of the difference is coming from.

Is that exactly what the output looks like? I'm seeing a temperature message intertwined in there: 

> Recv:
> Recv: 0.079 0.039 0.044 -0.023 0.014 -0.043 **T:18.39 /0.00 B:18.40 /0.00 @:0 B@:0**
> Recv: -0.011 -0.036 . .

If it's doing *that*, it actually sounds like a bug...and it'd make it particularly difficult to parse the mesh properly.

Gaara14, 2021-03-01:
Yes, this is the output in octoprint terminal.
I can of course suppress the temperature output, and the M420 output is clean, but still no graphical result.

The-EG, 2021-03-01:
Ok. It *could* just be how OctoPrint is displaying it if the suppress temps options hides it.

I'll try to parse that mesh without assuming it could having other data intermingled with it first, and we can see how that works. If it does have the temp data in the middle of the mesh (not just a line in the middle, but in the middle of a line like your first output) then that will be *much* more involved.

Gaara14, 2021-03-01:
For the first output, I have just copy/paste the raw output
Here is another try with the temperature display, if it can help :
Send: M420 V1 T1
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,9)                                                                   (9,9)
Recv: (5,295)                                                                (315,295)
Recv:      .          .          .          .          .          .          .          .          .    T:17.35 /0.00 B:16.52 /0.00 @:0 B@:0
Recv:        .
Recv: 
Recv:   0.059   -0.003   -0.011   -0.051    0.007   -0.011    0.004   -0.036       .          .
[etc]
Thank you for trying to fix this ! I hope it will work :)
If you want another info, don't hesitate to ask

The-EG, 2021-03-02:
Some notes: 
It looks like this firmware doesn't actually support the `T1` flag for `M20`. 

I'll see if I can add support, in general, for these cases but it's a bit more involved since the output without T1 is really meant to be human readable rather than for parsing.