The-EG/OctoPrint-UBLMeshEdit #10: [BUG] Get Mesh doesn't import mesh from terminal

opedog, 2021-03-08:
**Describe the bug**
I hit Get Mesh and nothing shows up

**M420 Output**
[image]
[image]

**Firmware**
ender 3 pro, marlin v2 i compiled myself in the last 2 weeks from th3d unified

opedog, 2021-03-08:
I'd be happy to provide any other information!

The-EG, 2021-03-08:
Hey, someone actually read the notice!

Ok, this *looks* like it's Mesh Bed Leveling (MBL) not Unified Bed Leveling (UBL) or ABL. It looks like it's similar enough to ABL that I can probably get it to pick it up, but the same caveats will apply (only basic functionality).

If it *is* MBL, I'd recommend trying out UBL. You can still use it without a probe (you need to also enable `#define PROBE_MANUALLY`. The process is similar, but you get the extra features of UBL like storing multiple meshes and it behaves much better (MBL treats a mesh with all 0's as invalid, etc.)

opedog, 2021-03-08:
OH CRAP I'm so sorry you're right this is MBL.  You can close this.  Mea culpa!

closed 2021-03-08 by The-EG