The-EG/OctoPrint-UBLMeshEdit #12: [BUG]  No Mesh

rpersak, 2021-04-06:
**Describe the bug**
 No Mesh
The command completed successfully, but no mesh was detected. If you are sure that you have UBL setup properly and are still getting this message, this may be a bug. Please report it: Create New Bug Report

**M420 Output**
Send: M420 V1 T1
Recv: Mesh Bed Level data:
Recv: 3x3 mesh. Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2
Recv:  0 +0.37500 +0.47500 +0.52500
Recv:  1 +0.45000 +0.50000 +0.52500
Recv:  2 +0.40000 +0.45000 +0.50000
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok

**Firmware**
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x

The-EG, 2021-04-06:
Hello, can you please verify that you are using ABL or UBL?

This looks like MBL, which I can't really support due to a few quirks in how it works (you can use UBL without a probe by enabling `PROBE_MANUALLY`).

rpersak, 2021-04-06:
Hey,

I'm using manual bed leveling with "#define MESH_BED_LEVELING" enabled...

I will try with now with probe_manually option...

rpersak, 2021-04-06:
OK, it's working now.

closed 2021-04-06 by rpersak