The-EG/OctoPrint-UBLMeshEdit #14: [BUG] plugin says No Mesh, but g29 returns mesh

whitewhidow, 2021-05-18:
**Describe the bug**
plugin says No Mesh, but g29 returns mesh:
```
Send: G29 S0
Recv: Mesh Bed Leveling ON
Recv: 3x3 mesh. Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2
Recv:  0 -0.22500 +0.02500 +0.07500
Recv:  1 -0.10000 +0.07500 -0.02500
Recv:  2 +0.05000 +0.12500 -0.02500
Recv: 
Recv: X:-5.00 Y:0.00 Z:0.27 E:184.88 Count X:-400 Y:0 Z:3
Recv: ok
```
Mesh was set using the following :
https://marlinfw.org/docs/gcode/G029-mbl.html

instead of using a probe
**M420 Output**
If this bug is related to the mesh not loading or updating properly, include the output of `M420 V1 T1` including the last 'ok'.

**Firmware**
What firmware (vanilla Marlin, etc), what version and is it a precompiled binary? If precompiled, where is it from? (Vendor, marlin.crc.in.au, etc.)

The-EG, 2021-05-19:
Hello,
I can see from the output:
> Recv: Mesh Bed Level data:

And the document you linked that you are using MBL not UBL. This plugin does not support MBL.

While the two are similar, MBL has some quirks that would make supporting it much more complex and convoluted. I recommend switching to UBL with `PROBE_MANUALLY` instead if you'd like to use this plugin.

closed 2021-05-19 by The-EG

whitewhidow, 2021-05-19:
> Hello,
> I can see from the output:
>
>
> And the document you linked that you are using MBL not UBL. This plugin does not support MBL.
>
> While the two are similar, MBL has some quirks that would make [...]

Could you explain this further ?   switching to UBL ?"    will this work the same way tho ?     wont that mess things up and make it think i have an actual probe ? 

thx in advance

whitewhidow, 2021-05-19:
or can i just purely change 
`#define MESH_BED_LEVELING` to `#define AUTO_BED_LEVELING_UBL`
will the procedure and lcd items etc remain the same ?       what is the actual difference between the 2 then ?    i thought one was manual and one was auto,   but seems you can do both manually ?    but then what is the actual diff ?

The-EG, 2021-05-19:
Yes, but you'd also need to `#define PROBE_MANUALLY` above in the probe section. It's not exactly the same as MBL, but the concept is the same and the commands are similar. There are some advantages to UBL, most notably a mesh that is all 0's isn't invalid and will display, and you can save multiple meshes to 'slots' that don't get erased when the EEPROM is reset.

I've detailed some of this on my blog: https://the-eg.github.io/2021/04/28/using-ubl-with-probe-manually.html
The Marlin documentation also has some details on UBL usage without a probe: https://marlinfw.org/docs/features/unified_bed_leveling.html#ubl-without-a-z-probe

whitewhidow, 2021-05-19:
ok i tried it,   but with the new options,    it disables ubl every time i start the machine ? 

and also,   the cld item that lets me manually choose a point,   and then move the z with the know,   jsut kleeps cycling between points,   i have no option of ever finishing this or going back to the previous menu ?

whitewhidow, 2021-05-19:
ok, so im at a point now,    where the plugin reports a zero mesh and the corner discs print properly 
but whenever i boot the machine, it will say UBL disabled,   nor do i know how to change this mesh via the lcd, without getting stuck ..

The-EG, 2021-05-19:
You'll have to enable UBL just like you would MBL (`M420 S`) unless you tell it to automatically enable with `G28.` 

I don't recall the exact defines, but there are additional options to enable LCD menus that I mention in the post I linked and the Marlin docs.

whitewhidow, 2021-05-19:
i have all the lcd menus enabled,   as thats how i did it earlier aswell,   i also have auto enable after g28 enabled,    but it will boot with it disabled, and after a g28 it will also still be disabled

if enabled,   it will re-enable after g28,   but if rebooted,   its disabled again