The-EG/OctoPrint-UBLMeshEdit #18: [FR] Try not to crash the printer by writing bad values

ali1234, 2024-06-12:
**Describe the new feature or functionality, and how it should work**
When editing the mesh, if you put an invalid Z value (ie one which is not a multiple of 0.05) or simply leave the Z field blank, the printer returns a M421 incorrect parameter error, and then Octoprint shuts it down with M311, requiring a power cycle. This is annoying.

Instead, the plugin should refuse to send these incorrect values, especially if the Z value field is completely blank.

**Firmware**
Marlin 2.1.3.2 - self-compiled.

ali1234, 2024-06-12:
Please don't do this:

[image]

[image]

ali1234, 2024-06-13:
It seems the only situation this happens is when I, J or Z are undefined, or in the case of Z an empty string. The firmware seems okay with anything that looks like a decimal, regardless of length.

The-EG, 2024-06-13:
Just tested myself and an arbitrary value, such as -0.173, works just fine with bufix-2.1.x. The web value will highlight it as invalid, but you can still submit it.

As for OctoPrint sending M112, this is configurable and an overly safe default. I would suggest turning it off it is annoying as **any** error will trigger that behavior.

ali1234, 2024-06-13:
There are a couple ways to reproduce it with undefined values:

 * Get mesh, click a square, type "hello" in Z field and then click save
 * Get mesh, click a square, reload the page, and then click save

I'm working on a patch to fix this.