The-EG/OctoPrint-UBLMeshEdit #19: [BUG] Z value of 0.025 is rounded to 0.03 in the interface

ali1234, 2024-06-12:
**Describe the bug**
Z value of 0.025 is rounded to 0.03 in the interface. 0.03 is not even a valid value - attempting to write it will cause the printer to respond with an error code (#18). But this bug is entirely in the display code - the correct value is written to the printer.

To reproduce:
1. Go to UBL editor
2. Click on a square
3. Type "0.025" in Z value box.
4. Click "Save value"
5. Look at the square.

Interface:
[image]

Actual data as seen by the printer:
```
Send: G29 S-1
Recv: echo:  G29 I999
Recv: echo:  M421 I0 J0 Z0.0000
Recv: echo:  M421 I0 J1 Z0.0250
Recv: echo:  M421 I0 J2 Z0.0500
Recv: echo:  M421 I0 J3 Z0.0250
Recv: echo:  M421 I0 J4 Z0.0000
Recv: echo:  M421 I1 J0 Z0.0250
Recv: echo:  M421 I1 J1 Z0.1500
Recv: echo:  M421 I1 J2 Z0.2000
Recv: echo:  M421 I1 J3 Z0.1500
Recv: echo:  M421 I1 J4 Z0.0250
Recv: echo:  M421 I2 J0 Z0.0500
Recv: echo:  M421 I2 J1 Z0.2000
Recv: echo:  M421 I2 J2 Z0.4000
Recv: echo:  M421 I2 J3 Z0.2000
Recv: echo:  M421 I2 J4 Z0.0500
Recv: echo:  M421 I3 J0 Z0.0250
Recv: echo:  M421 I3 J1 Z0.1500
Recv: echo:  M421 I3 J2 Z0.2000
Recv: echo:  M421 I3 J3 Z0.1500
Recv: echo:  M421 I3 J4 Z0.0250
Recv: echo:  M421 I4 J0 Z0.0000
Recv: echo:  M421 I4 J1 Z0.0250
Recv: echo:  M421 I4 J2 Z0.0500
Recv: echo:  M421 I4 J3 Z0.0250
Recv: echo:  M421 I4 J4 Z0.0000
```

ali1234, 2024-06-13:
In fact Marlin only stores the Z values to 2 decimal places, so rounding happens firmware side.

closed 2024-06-13 by ali1234

The-EG, 2024-06-13:
And you can even have as many decimals as you may want with `SERIAL_FLOAT_PRECISION` in configuration_adv.h